FreeCAD/FreeCAD_Conda #9: Drawing Export Page feature breaks on use on unicode()

alexpawlowski, 2017-05-25:
I'm not sure if this is the best venue, and I know FreeCAD on Conda is still a WIP, but I have been pretty successful in using FreeCAD on conda in a Windows environment, thanks for all of the effort to do this so far.

I noticed that when I attempt to save a drawing file as an svg using the Drawing toolbar's default export page feature, it breaks as the gui sends a command to use `unicode()` to instantiate the new svg file. 

Since it breaks there I can't find what the rest of the command is to execute the save while using Python 3's `open()` syntax. I saw that this might have been postulated in the [forum](https://forum.freecadweb.org/viewtopic.php?f=10&t=12534&start=380#p173983), so I just wanted to bring to your attention and see what I could use a workaround for now.

Here are my specs, I'll try updating to the latest release (I believe this was the one from a month ago)
```
OS: Windows 10 Enterprise
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.10700 (Git)
Build type: Release
Branch: py3-25
Python version: 3.5.3
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.0.0
```

looooo, 2017-05-26:
thanks for testing the package and reporting the issue.
currently we are moving python3 support into the master branch. I hope this is done soon. at the moment this is the most important step, as some developers ignore py3 quite a bit an bring in py3 incompabilities quite often.
after this step we will try to make our ci tools work with py3 to get automated testing. once this is done the py3 testing will start and I will try to help fix bugs like the one you have mentioned.
If you have some time left, you can report this problem on the freecad bugtracker. maybe there it will get a bit more attention...

alexpawlowski, 2017-05-26:
Okay, great thanks, I will put this up on the freecad bugtracker.

closed 2017-05-26 by alexpawlowski