FreeCAD/FreeCAD_Conda #13: Installing `freecad-launcher` package creates only empty folder

zwn, 2018-05-18:
Installing `freecad-launcher` package creates only empty folder `c:\Users\All Users\Microsoft\Windows\Start Menu\Programs\FreeCAD3 (64-bit)\` - no shortcut inside.

looooo, 2018-05-18:
I haven't tested this for a long time and the feedstock needs to be updated. If you have some time it would be nice if you could experiment with the freecad-launcher recipe:
https://github.com/FreeCAD/FreeCAD_Conda/tree/master/freecad_launcher

zwn, 2018-05-18:
I wanted to give it a shot but it seems anything I touch, breaks 😟 . See https://github.com/conda-forge/conda-build-feedstock/issues/51

looooo, 2018-05-18:
I have no problem with conda-build 3.10.3. You must install conda-build in the root environment. (The active environement when you first start anaconda-prompt)
how did you install conda-build?

zwn, 2018-05-18:
I did install it in the root environment (`conda install conda-build`). It is strange you have errors reported on installing conda-build 3.10.3 because I manually downloaded the package and verified the sha256 for those 3 files against `paths.json` and indeed they are wrong. They are also harmless since the files are not used by the python36 (conda-build probably works just fine).

Anyway, the problem has been that the first time I tried it with old conda. Now I am using the latest 4.5.4 and installing `freecad-launcher` creates the link to launch Freecad (so it was actually a bug in old conda). Only bummer is the cmd console that stays open until Freecad is closed again.

So, sorry for the noise.

closed 2018-05-18 by zwn