FreeCAD/FreeCAD_Conda #14: Cannot use with python3

gnthibault, 2018-05-23:
Dear developer,

Thank you for this nice project. I must admit that, although I have been struggling a lot in the past for compiling C++ projects with lots of dependencies, I felts even worse to try to compile Freecad and Pivy with QT5 and python3.

I thought I found the graal when I discovered this github, but I may be wrong, (please correct me in this case)

I tried to create a conda env with freecad and python 3.4:

```
conda create -n freecad freecad python=3.4
Solving environment: failed

UnsatisfiableError: The following specifications were found to be in conflict:
  - freecad -> netgen -> python=2.7
  - python=3.4
Use "conda info <package>" to see the dependencies for each package.
```

What can I do to correct for this ?
I tried previously without specifying the python version, but I ended up with an error in a Pivy call that complained about missing Pyside.

Then my attempt at pip install Pyside resulted in an error because Pyside is not compatible with my python version, which is 3.6.

looooo, 2018-05-23:
the only supported version of python is 3.6! I won't support any older version. building for 3 plattforms is alteady way too much work. so removing the python constraint should solve the issue.

> Versionen, but I ended up with an error in a Pivy call that complained about missing Pyside.

can you explain?

gnthibault, 2018-05-23:
Yes sorry, here is my workflow:

conda create -n freecad freecad netgen=6.2.1804=py36_0
[everything works like a charm]

Then I try to run my software:
(freecad):~/projects/freecadScope$ python ./SimulatorGui.py 
Traceback (most recent call last):
  File "./SimulatorGui.py", line 32, in <module>
    from pivy.quarter import QuarterWidget
  File "/home/user/anaconda3/envs/freecad/lib/python3.6/site-packages/pivy/quarter/__init__.py", line 17, in <module>
    from . import QuarterWidget
  File "/home/user/anaconda3/envs/freecad/lib/python3.6/site-packages/pivy/quarter/QuarterWidget.py", line 113, in <module>
    from PySide import QtCore, QtGui, QtOpenGL
ModuleNotFoundError: No module named 'PySide'
(freecad) ~/projects/freecadScope$ which pip
/home/gnthibault/anaconda3/envs/freecad/bin/pip
(freecad) :~/projects/freecadScope$ pip install pyside
Collecting pyside
  Using cached https://files.pythonhosted.org/packages/36/ac/ca31db6f2225844d37a41b10615c3d371587677efd074db29855e7035de6/PySide-1.2.4.tar.gz
    Complete output from command python setup.py egg_info:
    only these python versions are supported: [(2, 6), (2, 7), (3, 2), (3, 3), (3, 4)]

gnthibault, 2018-05-23:
I think I should find a way of using this version of Pivy: https://github.com/FreeCAD/pivy/tree/master/pivy
Instead of the one featured in the conda env, no ? is it easy to do ?

looooo, 2018-05-24:
> EDIT: It looks like conda install pivy=0.6.4b2 solved my problem.

Yes you did it the right way. I now disabled all pivy packages from the freecad-channel, so pivy should be pulled automatically from the conda-forge. This will pull the same package as you have installed with specifying the version.

> Edit2: as a workaroud, I did something like this:
FREECADPATH = '/home/user/anaconda3/envs/freecad/lib/'
sys.path.append(FREECADPATH)

You can use multiple versions of FreeCAD in one environment. This is done for example in ubuntu where you can install daily-builds and version x.y side by side. This is only possible without adding the package to the python-path directly. Therefor you have to manipulate the path before FreeCAD gets importable.

You can also use some conda-env-variables:
```
import os, sys
sys.path.append(`os.path.join(os.environ["CONDA_PREFIX"], "lib")
```

Multiple versions of freecad in one conda-env doesn't make sense at all. I already thought about adding a meta-package which handles the path-problem in a more pythonic way...

gnthibault, 2018-05-24:
Thank you very much for your help, and the work you have done maintaining FreeCAD and its related tool. Without your work I would probably not have been able to use FreeCAD scripting capability at all.

closed 2018-05-24 by gnthibault

selvakarna, 2019-10-03:
So , which version Support FreeCAD? 

 can you share clear link to install FreeCAD in Python3.6