FreeCAD/FreeCAD_Conda #16: OpenSCADUtils.py needs syntax edits to be used with Python3.

Lordling, 2018-07-10:
So we managed to sew FreeCAD and its related dependencies into our environment and got it working with Python3 (OS Ubuntu 18.04, Bionic Beaver). However, I had to manually edit and update OpenSCADUtils.py in a few places in order to use the mesh difference function without crashing. 

Namely, a couple of generator objects [g] still use the Python2 g.next() syntax instead of the Python3 next(g) [or g.__next__() according to some light googling]. Specifically for the OpenSCADUtils meshoptempfile functions, I had to make the following two changes to make my code useable (line numbers included in images):

[image]
[image]

The changes are relatively minor but it's something that we can't track in our repo itself, FreeCAD being an external library and all. Would it be possible to get an updated OpenSCADUtils.py officially supported for the purpose of using meshoptempfile in Python3 applications?

looooo, 2018-07-10:
thanks for reporting. PR is on it's way. If you find any other problems please report them to https://forum.freecadweb.org/viewtopic.php?f=10&t=12534&start=660 or if possible create a PR for freecad directly.

looooo, 2018-11-03:
@Lordling have you tested this in a recent version of FreeCAD. I guess it should be fixed. If so please close the issue.

closed 2018-11-12 by Lordling