FreeCAD/FreeCAD_Conda #17: Menu blanck

Mx74, 2018-07-13:
Hello,

Having follow the installation instruction on Ubuntu 18.04, everything seems to work, Freecad launches well, but every menu is grey and unreadable.
Mouse contextual menu, or any menu of the interface, no difference.
Everything works, you can click and have actions done But just can't red what you're doing.

looooo, 2018-07-13:
It's an issue with qt5.6 and should go away with qt5.9. But qt5.9 is not yet available as a package @ conda-forge.

can you post the errors reported in the console? I remember setting some environment-variables to make it work. something to force gtk2 style, But I can't find at the moment.

Mx74, 2018-07-13:
Can't access the console, as I can't red the menu. 
And the terminal shows no error.

Mx74, 2018-07-13:
Ok, I could access it. No error shown...

looooo, 2018-07-13:
found it:
`export QT_STYLE_OVERRIDE=gtk2`
ref: https://stackoverflow.com/questions/50555973/qgtkstyle-could-not-resolve-gtk

Mx74, 2018-07-17:
Worked, many thanx again

closed 2018-07-17 by Mx74

looooo, 2018-11-03:
@Mx74 if you have an updated ubuntu 18.04, this workaround is not necessary any-more.

Mx74, 2018-11-07:
Ok, thank you for hint. 
No I haven't upgraded yet, as everything works now, I didn't take the risk ;-)
I'll try later when I'll hve more time