FreeCAD/FreeCAD_Conda #18: Check out linuxdeploy-plugin-conda

probonopd, 2018-08-30:
Maybe you are interested in https://github.com/TheAssassin/linuxdeploy-plugin-conda#example which could potentially simplify AppImage generation from Conda packages a lot.

looooo, 2018-11-03:
thanks for the information. Actually our script to create the appimage is pretty simple, but includes some manually steps (removing unnecessary stuff, some workarounds...) It can be found here:
https://github.com/FreeCAD/FreeCAD-AppImage/blob/master/conda/conda.sh

We using this now for several months to make py3/qt5 builds of freecad available. I am very happy with this solution as it is very simple compared to using conda directly...

TheAssassin, 2019-05-21:
Fun fact: the idea to make a conda plugin was partially inspired by you using miniconda.