FreeCAD/FreeCAD_Conda #19: OpenSCAD Executable reporting as unavailable in spite of being installed + present.

Lordling, 2018-11-12:
Running on OS Ubuntu 18.04 machine; I have FreeCAD and OpenSCAD both installed, with the openscad executable being in /usr/bin/ directory (confirmed by running 'which openscad' as well as, well navigation). OpenSCAD also run when called as a command from the terminal. Yet, consistently, OpenSCADUtils.py keeps raising an error that the OpenSCAD executable is unavailable.

I actually went to the searchforopenscadexe function, and ran the code myself in Python terminal; the code does in fact return '/usr/bin/openscad' as the path to executable. So the executable is installed, is functional, and is findable by the OpenSCADUtils' own code ... yet, it is generating OpenSCADError that executable is unavailable. 

At this point I'm simply flummoxed and unsure how to proceed. Any help would be appreciated.

I will say that this set-up with OpenSCAD and FreeCAD does work on a different machine of mine (also Ubuntu 18.04) but trying to replicate the set-up on a different machine is what is causing this rather arcane issue. Same app, same version of FreeCAD and OpenSCAD, same OS, different machine, weird issue of OpenSCAD executable simply not registering.

Bougakov, 2020-11-22:
I am getting the same issue on MacOS. Both FreeCAD and OpenSCAD are installed from Homebrew.

```
➜  ~ brew cask install openscad 
==> Downloading https://files.openscad.org/OpenSCAD-2019.05.dmg
Already downloaded: /Users/sanja/Library/Caches/Homebrew/downloads/e3b71d2b85d361ddac64368ece57f9615b0c6ac34849b98b155ea4417728f1fb--OpenSCAD-2019.05.dmg
==> Verifying SHA-256 checksum for Cask 'openscad'.
==> Installing Cask openscad
==> Moving App 'OpenSCAD.app' to '/Applications/OpenSCAD.app'.
🍺  openscad was successfully installed!
➜  ~ brew cask install freecad 
Updated 2 taps (homebrew/core and homebrew/cask).
==> Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/Free
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws
######################################################################## 100.0%
==> No SHA-256 checksum defined for Cask 'freecad', skipping verification.
==> Installing Cask freecad
==> Moving App 'FreeCAD.app' to '/Applications/FreeCAD.app'.
🍺  freecad was successfully installed!
```

Bougakov, 2020-11-22:
I figured it out. The error on opening CSG file generated by OpenSCAD was:

```
~ /Applications/FreeCAD.app/Contents/MacOS/FreeCAD -l
FreeCAD 0.18, Libs: 0.18R16146 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

End processing CSG file
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADFeatures.py", line 467, in execute
    fp.Operation, maxmeshpoints=maxmeshpoints)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 580, in process_ObjectsViaOpenSCADShape
    return process3D_ObjectsViaOpenSCADShape(children,name,maxmeshpoints)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 555, in process3D_ObjectsViaOpenSCADShape
    stlmesh = meshoptempfile(Operation,meshes)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 456, in meshoptempfile
    result = callopenscadmeshstring('%s(){%s}' % (opname,meshimports))
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 420, in callopenscadmeshstring
    tmpfilename=callopenscadstring(scadstr,'stl')
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 198, in callopenscadstring
    keepname=True)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in callopenscad
    raise OpenSCADError('OpenSCAD executable unavailable')
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
Exception (Sun Nov 22 21:20:50 2020): Input shape is null  
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADFeatures.py", line 467, in execute
    fp.Operation, maxmeshpoints=maxmeshpoints)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 580, in process_ObjectsViaOpenSCADShape
    return process3D_ObjectsViaOpenSCADShape(children,name,maxmeshpoints)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 555, in process3D_ObjectsViaOpenSCADShape
    stlmesh = meshoptempfile(Operation,meshes)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 456, in meshoptempfile
    result = callopenscadmeshstring('%s(){%s}' % (opname,meshimports))
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 420, in callopenscadmeshstring
    tmpfilename=callopenscadstring(scadstr,'stl')
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 198, in callopenscadstring
    keepname=True)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py", line 182, in callopenscad
    raise OpenSCADError('OpenSCAD executable unavailable')
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
Exception (Sun Nov 22 21:20:51 2020): Input shape is null  
```

The problematic code in `/Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD/OpenSCADUtils.py` was:

```
    osfilename = FreeCAD.ParamGet(\
        "User parameter:BaseApp/Preferences/Mod/OpenSCAD").\
        GetString('openscadexecutable')
```

It referenced the `~/Library/Preferences/FreeCAD/user.cfg` file that was missing the text inside the `<FCParamGroup Name="OpenSCAD" />` tag.

I changed it to:

```
        <FCParamGroup Name="Mod">
          <FCParamGroup Name="OpenSCAD">
            <FCText Name="openscadexecutable">/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD</FCText>
          </FCParamGroup>
```

And this solved the issue.

There was no other way to tell FreeCAD how to find OpenSCAD executable in the GUI.

Passinir, 2021-02-11:
Thank you Bougakov, it solved the issue for me too. 
On Windows 10 with FreeCAD 0.19 I found the file in 
C:\Users\<myname>\AppData\Roaming\FreeCAD\user.cfg 
and after the line
<FCParamGroup Name="OpenSCAD">
added the line
<FCText Name="openscadexecutable">C:/Program Files/OpenSCAD/openscad.exe</FCText>

Alternatively it can be done in the menu Tools - Parameter Editor adding a New String Item in BaseApp\Preferences\Mod\OpenSCAD with name openscadexecutable and content C:/Program Files/OpenSCAD/openscad.exe

chennes, 2021-10-21:
That preference is accessible in the GUI via Preferences->OpenSCAD->OpenSCAD Executable -- note that you'll have to load the OpenSCAD WB first, either using the Workbenches preferences panel, or by selecting it in the menu.

Erudition, 2021-12-15:
I'm using the Flatpak versions of both FreeCAD and OpenSCAD. 
Even when I find the `openscad` executable and put the path in the preferences, FreeCAD says the executable is unavailable. 
Enabled all filesystem permissions for FreeCAD in Flatseal, too, to no avail.

luzpaz, 2022-02-07:
CC @hfiguiere
Hi! Just wondiering if you've heard of issues involving OpenSCAD on the FreeCAD flatpak as well?

hfiguiere, 2022-02-07:
if you install the openscad flatpak it should just work.

https://flathub.org/apps/details/org.openscad.OpenSCAD

hfiguiere, 2022-02-07:
if you change the executable to something in `/usr/bin` it's wrong.