FreeCAD/FreeCAD_Conda #25: Add VTK info to README Packaging Status section

luzpaz, 2019-06-17:
I don't see VTK in the [README Packaging Status](https://github.com/FreeCAD/FreeCAD_Conda#packaging-status) table but I do see it as a submodule in this repo. Should I add it ? Looks like VTK feedstock is at v8.2.0

luzpaz, 2019-06-17:
also per your reply in https://github.com/FreeCAD/FreeCAD_Conda/issues/10#issuecomment-502683540 it seems like the Conda packaging is not pinned to 8.1.0 anymore but in fact using 8.2.0.

looooo, 2019-06-17:
yes, I think it would be ok to have it listed, allthough I am not a maintainer of vtk. For conda-forge we haven't yet updated to vtk8.2 (still some cmake problems) But the weekly builds (freecad/label/dev-channel) should use vtk8.2.0.

luzpaz, 2019-06-19:
@looooo 2 quick questions 
1. Where is the list of minimum versions for all FreeCAD dependencies?  
2. Where are all the conda versions listed for all FreeCAD dependencies?

looooo, 2019-06-19:
1.:
- search for freecad (anaconda.org)
https://anaconda.org/search?q=freecad
- select channel
eg.: https://anaconda.org/conda-forge/freecad
- click on files
- click on the info (i) to get information of a package. This gives also a list of dependency-versions.
eg.: 
```
boost-cpp >=1.68.0,<1.68.1.0a0, coin3d, freetype >=2.9.1,<3.0a0, hdf5 >=1.10.4,<1.10.5.0a0, libgcc-ng >=7.3.0, libstdcxx-ng >=7.3.0, netgen, numpy, occt >=7.3.0,<7.4.0a0, pivy, pyside2 >=5.6.0a.1,<6.0a0, python >=3.7,<3.8.0a0, pyyaml, qt >=5.6.2,<5.7.0a0, six, smesh, tbb >=2019.7, vtk >=8.1.2,<9.0a0, xerces-c >=3.2.2,<3.2.3.0a0, xorg-libxi, xorg-libxmu, xorg-libxt, zlib >=1.2.11,<1.3.0a0
```

2. What do you mean with conda-version?

luzpaz, 2019-06-19:
i'm asking about minimum versions for freecad dependencies (even if they are not conda installed).

luzpaz, 2019-06-19:
@looooo thanks. Is there an automated way to get the list you mentioned?

looooo, 2019-06-19:
> i'm asking about minimum versions for freecad dependencies (even if they are not conda installed).

I guess there is no such list.

> @looooo thanks. Is there an automated way to get the list you mentioned?

there must be, but I have no clue how to get it.

luzpaz, 2019-09-08:
vtk was added a while ago .. closing issue

closed 2019-09-08 by luzpaz