FreeCAD/FreeCAD_Conda #26: FEM Workbench not available

GitWyd, 2019-06-25:
Hi,

After installing FreeCAD as instructed and successfully starting it, I am unable to access the FEM workbench. Instead I am presented with the following error message:

libvtkIOXML-8.1.so.1: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "<string>", line 45, in Initialize

Do you have an idea on how I can fix this?

Best,
Philippe

looooo, 2019-06-26:
Vtk wasn't pinned the right way. Please install freecad from freecad/label/dev (0.19pre) or install a compatible vtk:

`conda create -n freecad freecad qt=5.9.* -c freecad/label/def -c conda-forge`
`conda create -n freecad freecad vtk=8.1.* -c comda-forge`

Hopefully this helps solving this issue.

GitWyd, 2019-06-26:
Thank you for your quick response.

`conda create -n freecad freecad qt=5.9.* -c freecad/label/def -c conda-forge`
The above approach yielded the following error:
```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

  - freecad -> pivy -> pyside2 -> qt[version='>=5.9.7,<5.10.0a0']
  - qt=5.9
```
The second line worked:
` conda create -n freecad freecad vtk=8.1.* -c conda-forge`

However, when I try to start FreeCAD in the newly created environment, I get the following issues at startup:```

FreeCAD 0.18, Libs: 0.18R16117 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Gtk-Message: 09:37:27.607: Failed to load module "atk-bridge"
Gtk-Message: 09:37:27.610: Failed to load module "canberra-gtk-module"
```
Any suggestions?

looooo, 2019-06-26:
sry, I made one mistake. It's not freecad/label/def but freecad/label/dev. I hope this solves this issue.

hex41434, 2021-02-18:
Hi, I created an env using this command : 

conda create -n freecad freecad qt=5.9.* -c freecad/label/def -c conda-forge 

and I run one old FEM Macro using FreeCAD gui , in which I had CalculiX as solver, and I got this error : 

FEM: CalculiX binary ccx not found in standard system binary path. Please install ccx or set path to binary in FEM preferences tab CalculiX.

could you please guide me how can I fix this? ( I tried it on both Mac and Windows)

Thanks in advance

looooo, 2021-02-18:
You can install calculix via `conda install calculix -c conda-forge`

hex41434, 2021-02-18:
thanks a lot for your fast reply. 

my problem is solved in windows, but in Mac I got this:

Running the Python command 'FEM_SolverRun' failed:
Traceback (most recent call last):
  File "/Applications/anaconda3/envs/freecad_fem_env/Mod/Fem/femcommands/commands.py", line 1022, in Activated
    run_fem_solver(self.selobj)
  File "/Applications/anaconda3/envs/freecad_fem_env/Mod/Fem/femsolver/run.py", line 65, in run_fem_solver
    fea.run()
  File "/Applications/anaconda3/envs/freecad_fem_env/Mod/Fem/femtools/ccxtools.py", line 976, in run
    ret_code = self.ccx_run()
  File "/Applications/anaconda3/envs/freecad_fem_env/Mod/Fem/femtools/ccxtools.py", line 912, in ccx_run
    progress_bar = FreeCAD.Base.ProgressIndicator()

cannot create 'N4Base19ProgressIndicatorPyE' instances

hex41434, 2021-02-18:
as I did not need the progress bar, I commented it in ccxtool.py and now it works! ( I am not sure if it is a good solution though!)

mnesarco, 2022-12-14:
Hello, I am facing the same problem in macos. Is there any known solution to this? 
```
cannot create 'N4Base19ProgressIndicatorPyE' instances
```

wwmayer, 2023-01-04:
What happens when you enter this into FreeCAD's Python console?
`progress_bar = FreeCAD.Base.ProgressIndicator()`

Does this already raise this error?

mnesarco, 2023-01-04:
As a temporary workaround in MarzWorkbench, I have faked the ProgressIndicator for Mac:
https://github.com/mnesarco/MarzWorkbench/blob/c8072ec829ce12c397e977fcaffc68206bfcd4b6/freecad/marz/extension/ui.py#L61

wrenchInTheWorks, 2023-05-19:
> What happens when you enter this into FreeCAD's Python console? `progress_bar = FreeCAD.Base.ProgressIndicator()`
> 
> Does this already raise this error?

Greetings, I'm also having this error and I tested as you say and it sure does! Any idea what's causing this? I'm also on MACos. Application was running fine yesterday, tired to import a DXF today and this happened. Any help would be much appreciated.

wwmayer, 2023-05-20:
With the MSYS build I had the same problem and it was fixed with: https://github.com/FreeCAD/FreeCAD/pull/8888