FreeCAD/FreeCAD_Conda #37: No module named 'WebGui'

incal, 2020-03-28:
Hello,
I followed the whole installation procedure on Macos Sierra, but I have this problem

`FreeCAD 0.18, Libs: 0.18R16146 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

No module named 'WebGui'`

looooo, 2020-03-28:
please install freecad (whatever version) from the `freecad/label/dev` channel if you care about the webmodule. For using freecad it's not mandatory to have the webmodule available.

Anyway this issue will be solved with 0.19 release.

incal, 2020-03-29:
hi,
thank's for reply
unfortunately I break may conda installation:

`conda deactivate freecad_test
conda env remove -n  freecad_test
conda config --add channels freecad/label/dev
conda create -n freecad_test freecad

➜  ~ conda create -n freecad_test freecad
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                         

UnsatisfiableError: 
`
can you help me?

looooo, 2020-03-29:
interesting. I don't see a reason why it should fail. Is there some additional information printed?

incal, 2020-03-29:
i am new of conda environment
how can I provide more information about it?

looooo, 2020-03-29:
Normally there should be a list of conflicting packages if conda create fails.

Btw. we are running the conda create command daily to create the freecad bundles for (win, linux, osx): https://github.com/FreeCAD/FreeCAD-AppImage/blob/master/conda/osx_dev/osx_dev.sh#L2L11

and this was working today, so I don't think this is a general issue.

You can also try to create an environment with one of the base dependencies like pivy, occt, boost and see if this also leads to conflicts. If it does try to do the same with one of them to see which one leads to conflicts.

`conda create -n test pivy occt boost`

incal, 2020-03-29:
` conda create -n test pivy occt boost
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - occt

Current channels:

  - https://conda.anaconda.org/freecad/label/dev/osx-64
  - https://conda.anaconda.org/freecad/label/dev/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.`

looooo, 2020-03-29:
please post the output of:

```
conda config --show channels
```

maybe there is the `conda-forge` channel missing?

incal, 2020-03-29:
Yes you are right
I must have done something wrong

now works!

but the main menu does not respond to the click, because it is a development version?

incal, 2020-03-29:
do you know why ai cannot use Inventor Loader module?

`Sono stati trovati alcuni errori che impediscono di installare questo ambiente di lavoro: Modulo python mancante: xlutils, Modulo python mancante: xlrd, Modulo python mancante: xlwt, Modulo python mancante: olefile , . Si prega di installare prima i componenti mancanti.`

looooo, 2020-03-29:
try to install the missing modules?
`conda install xlutils xlrd xlwt olefile`

incal, 2020-03-29:
`(freecad_test) ➜  ~ conda install xlutils xlrd xlwt olefile
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                         

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - olefile -> python[version='2.7.*|3.5.*|3.6.*|3.4.*|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=2.7,<2.8.0a0']
  - xlrd -> python[version='2.7.*|3.5.*|3.6.*|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=2.7,<2.8.0a0|>=3.5,<3.6.0a0']
  - xlutils -> python[version='>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']

Your python: python=3.8

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package bzip2 conflicts for:
xlrd -> python -> bzip2[version='>=1.0.6,<2.0a0|>=1.0.8,<2.0a0']
olefile -> python -> bzip2[version='>=1.0.6,<2.0a0|>=1.0.8,<2.0a0']
xlutils -> python[version='>=3.6,<3.7.0a0'] -> bzip2[version='>=1.0.6,<2.0a0|>=1.0.8,<2.0a0']
xlwt -> python -> bzip2[version='>=1.0.6,<2.0a0|>=1.0.8,<2.0a0']
`

looooo, 2020-03-29:
Seems like one of those packages conflicts with installed versions. Again try to Install one by one.

incal, 2020-03-29:
thank you very much,
your help is very precious to me, I have been trying to do this for days ..
it is the xlutils package that has problems, I have installed the others

`(freecad_test) ➜  ~ conda install xlutils       
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                         

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - xlutils -> python[version='>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']

Your python: python=3.8

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

`

looooo, 2020-03-29:
xlutils isn't available on conda-forge. But you can install it with pip (the conda-env where you want the package must be activated):
`pip install xlutils`

I also uploaded a conda-package of xlwt to freecad/label/dev. So now it should also work to install with conda:
`conda install xlutils -c freecad/label/dev`

looooo, 2020-03-29:
https://github.com/conda-forge/staged-recipes/pull/11159

incal, 2020-03-29:
great!, you are an angel
I have successfully installed xlutils via the command:
`conda install xlutils -c freecad/label/dev`

something still goes wrong with the module:
`Traceback (most recent call last):
  File "/Users/incal/miniconda3/envs/freecad_test/Mod/AddonManager/addonmanager_workers.py", line 645, in run
    repo = git.Repo.clone_from(self.repos[idx][1], clonedir, branch='master')
AttributeError: module 'git' has no attribute 'Repo'
`
but I presume that you can't solve this problem

looooo, 2020-03-30:
https://tracker.freecadweb.org/view.php?id=4072

Maybe you can post the output of `conda list` with the `freecad_test` environment activated.

incal, 2020-03-30:
hi,
 I have already tried to install git but nothing change:
`(freecad_test) ➜  ~ conda install git
Collecting package metadata (current_repodata.json): done
Solving environment: done

# All requested packages already installed.
`

this is my conda list:

`(freecad_test) ➜  ~  conda list                                                                                                                                   
# packages in environment at /Users/incal/miniconda3/envs/freecad_test:
#
# Name                    Version                   Build  Channel
boost-cpp                 1.72.0               hdf9ef73_0    conda-forge
bzip2                     1.0.8                h0b31af3_2    conda-forge
ca-certificates           2019.11.28           hecc5488_0    conda-forge
certifi                   2019.11.28       py38h32f6830_1    conda-forge
coin3d                    4.0.0             h3c06c15_1008    conda-forge
curl                      7.68.0               h8754def_0    conda-forge
expat                     2.2.9                h4a8c4bd_2    conda-forge
flann                     1.9.1             h3c234d5_1006    conda-forge
fontconfig                2.13.1            h6b1039f_1001    conda-forge
freecad                   0.19.pre        py38h7f341c1_110    freecad/label/dev
freetype                  2.10.1               h8da9a1a_0    conda-forge
future                    0.18.2           py38h32f6830_1    conda-forge
gettext                   0.19.8.1          h46ab8bc_1002    conda-forge
git                       2.26.0          pl526h561ab23_0    conda-forge
hdf4                      4.2.13            h84186c3_1003    conda-forge
hdf5                      1.10.5          nompi_h3e39495_1104    conda-forge
icu                       64.2                 h6de7cb9_1    conda-forge
jpeg                      9c                h1de35cc_1001    conda-forge
jsoncpp                   1.8.4             ha1b3eb9_1002    conda-forge
krb5                      1.16.4               h1752a42_0    conda-forge
libblas                   3.8.0               16_openblas    conda-forge
libcblas                  3.8.0               16_openblas    conda-forge
libclang                  9.0.1           default_hf57f61e_0    conda-forge
libcurl                   7.68.0               h709d2b2_0    conda-forge
libcxx                    9.0.1                         1    conda-forge
libedit                   3.1.20170329      hcfe32e1_1001    conda-forge
libffi                    3.2.1             h4a8c4bd_1007    conda-forge
libgfortran               4.0.0                         2    conda-forge
libiconv                  1.15              h0b31af3_1006    conda-forge
liblapack                 3.8.0               16_openblas    conda-forge
libllvm9                  9.0.1                ha1b3eb9_0    conda-forge
libnetcdf                 4.7.4           nompi_he461dc0_100    conda-forge
libopenblas               0.3.9                h3d69b6c_0    conda-forge
libpng                    1.6.37               hbbe82c9_1    conda-forge
libpq                     12.2                 h554dc5a_0    conda-forge
libssh2                   1.8.2                hcdc9a53_2    conda-forge
libtiff                   4.1.0                h2ae36a8_6    conda-forge
libwebp-base              1.1.0                h0b31af3_3    conda-forge
libxml2                   2.9.10               h53d96d6_0    conda-forge
libxslt                   1.1.33               h320ff13_0    conda-forge
llvm-openmp               9.0.1                h28b9765_2    conda-forge
lz4-c                     1.8.3             h6de7cb9_1001    conda-forge
ncurses                   6.1               h0a44026_1002    conda-forge
netgen                    6.2.1808        py38h65d59e6_1007    conda-forge
nspr                      4.20              h0a44026_1000    conda-forge
nss                       3.47                 hc0980d9_0    conda-forge
numpy                     1.18.1           py38h1f821a2_1    conda-forge
occt                      7.4.0                hb9b6dc7_3    conda-forge
olefile                   0.46                       py_0    conda-forge
openssl                   1.1.1e               h0b31af3_0    conda-forge
pcl                       1.9.1             h3d0f52c_1005    conda-forge
pcre                      8.44                 h4a8c4bd_0    conda-forge
perl                      5.26.2            haec8ef5_1006    conda-forge
pip                       20.0.2                     py_2    conda-forge
pivy                      0.6.6dev         py38hd8f0d6a_0    freecad/label/dev
[... 1 lines cut ...]
pthread-stubs             0.4               h1de35cc_1001    conda-forge
pyside2                   5.13.2           py38ha8f7116_2    conda-forge
python                    3.8.2           hd5f0129_5_cpython    conda-forge
python_abi                3.8                      1_cp38    conda-forge
pyyaml                    5.3.1            py38h64e0658_0    conda-forge
qhull                     2015.2            h04f5b5a_1000    conda-forge
qt                        5.12.5               h514805e_3    conda-forge
readline                  8.0                  hcfe32e1_0    conda-forge
setuptools                46.1.3           py38h32f6830_0    conda-forge
six                       1.14.0                     py_1    conda-forge
smesh                     8.3.0.3          py38h62d2a6e_8    conda-forge
[... cut ...]
xlrd                      1.2.0                      py_0    conda-forge
xlutils                   2.0.0                      py_0    freecad/label/dev
xlwt                      1.3.0                      py_1    conda-forge
xz                        5.2.4             h0b31af3_1002    conda-forge
yaml                      0.2.2                h0b31af3_1    conda-forge
zlib                      1.2.11            h0b31af3_1006    conda-forge
zstd                      1.4.4                hed8d7c8_2    conda-forge`

looooo, 2020-03-30:
As it seems gitpython is not installed with conda, it's possible your system installation of gitpython is used and leads to this conflict as it might be an incompatible version.
`conda install gitpython`

incal, 2020-03-30:
wonderful, it's works!!
thank you very much

best regards

closed 2020-03-30 by incal