FreeCAD/FreeCAD_Conda #38: Build FreeCAD locally using Conda

amrit3701, 2020-04-30:
I am building FreeCAD locally using conda by following tutorial (https://wiki.freecadweb.org/Python_3) in Alternate Install section. 

[image]

In the instructions, you said to go `FreeCAD_Conda/.FreeCAD_debug` and set variable values. Here I unable to find `.FreeCAD_debug`.

Also, in [develop.md](https://github.com/FreeCAD/FreeCAD_Conda/blob/master/develop.md) there are

> Go to FreeCAD_Conda/.FreeCAD_debug/build.sh and modify the FREECAD_SOURCE variable at the top of the document.

Please let me know is these tutorials are outdated? Or I am going in wrong direction.

My goal is, I want to build FreeCAD locally using conda and then create a *.AppImage* using https://wiki.freecadweb.org/AppImage (Repackaging AppImages).

looooo, 2020-05-01:
sry, docs are a bit outdated.

If you want to build the appimage locally you need to do the following:
1. clone freecad-feedstock https://github.com/looooo/freecad-feedstock
2. change these lines to point to your local fork of freecad: https://github.com/looooo/freecad-feedstock/blob/master/recipe/meta.yaml#L10L11
See here how to specify the paths: https://docs.conda.io/projects/conda-build/en/latest/resources/define-metadata.html#source-section
3. build the recipe: `conda build . -m .ci_support_<platform/python>.yaml` (if this succeeds a package is stored in miniconda/conda-bld/linux64
4. clone FreeCAD/Appimage (+ Download AppImageKit https://github.com/AppImage/AppImageKit/releases and place it in the root Appimage directory (not really sure about this, maybe it's also stored one directory up)
5. add `--use-local` to conda command here: https://github.com/FreeCAD/FreeCAD-AppImage/blob/master/conda/linux_dev/linux_dev.sh#L2L11
6. `bash linux_dev.sh `

output should be an appimage of your local build.

amrit3701, 2020-05-05:
Thanks @looooo I will try this weekend and let you know the status.

amrit3701, 2020-05-08:
Hi @looooo 

I just installed a fresh LUbuntu on my Macbook pro and install miniconda (python3.7 support) (https://docs.conda.io/en/latest/miniconda.html). 

### System details

```bash
ambu@ambu-pc:~/Desktop/freecad_build/freecad-feedstock$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.04
Release:        19.04
Codename:       disco
```

Getting below error on running `conda build . -m .ci_support/linux_python3.8.____cpython.yaml` command:

Logs:

```bash
ambu@ambu-pc:~/Desktop/freecad_build/freecad-feedstock$ tree
.
├── azure-pipelines.yml
├── build-locally.py
├── conda-forge.yml
├── LICENSE.txt
├── README.md
└── recipe
    ├── bld.bat
    ├── build.sh
    ├── conda_build_config.yaml
    ├── cross-linux.cmake
    ├── meta.yaml
    └── yum_requirements.txt

1 directory, 11 files

ambu@ambu-pc:~/Desktop/freecad_build/freecad-feedstock$ conda build . -m .ci_support/linux_python3.8.____cpython.yaml 
No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
WARNING:conda_build.metadata:No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
Adding in variants from internal_defaults
INFO:conda_build.variants:Adding in variants from internal_defaults
Adding in variants from /home/ambu/Desktop/freecad_build/freecad-feedstock/recipe/conda_build_config.yaml
INFO:conda_build.variants:Adding in variants from /home/ambu/Desktop/freecad_build/freecad-feedstock/recipe/conda_build_config.yaml
Adding in variants from .ci_support/linux_python3.8.____cpython.yaml
INFO:conda_build.variants:Adding in variants from .ci_support/linux_python3.8.____cpython.yaml
Attempting to finalize metadata for freecad
INFO:conda_build.metadata:Attempting to finalize metadata for freecad
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed

Leaving build/test directories:
  Work:
 /home/ambu/miniconda3/conda-bld/work 
  Test:
 /home/ambu/miniconda3/conda-bld/test_tmp 
Leaving build/test environments:
  Test:
source activate  /home/ambu/miniconda3/conda-bld/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_p 
  Build:
source activate  /home/ambu/miniconda3/conda-bld/_build_env 

Traceback (most recent call last):
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/environ.py", line 757, in get_install_actions
    actions = install_actions(prefix, index, specs, force=True)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda/common/io.py", line 88, in decorated
    return f(*args, **kwds)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda/plan.py", line 474, in install_actions
    txn = solver.solve_for_transaction(prune=prune, ignore_pinned=not pinned)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda/core/solve.py", line 117, in solve_for_transaction
[... 53 lines cut ...]
    bypass_env_check=bypass_env_check)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/metadata.py", line 940, in parse_again
    bypass_env_check=bypass_env_check),
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/metadata.py", line 1528, in _get_contents
    rendered = template.render(environment=env)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "/home/ambu/Desktop/freecad_build/freecad-feedstock/recipe/meta.yaml", line 61, in top-level template code
    run:
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/jinja_context.py", line 230, in pin_compatible
    pins, _, _ = get_env_dependencies(m, 'host', m.config.variant)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/render.py", line 154, in get_env_dependencies
    channel_urls=tuple(m.config.channel_urls))
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/environ.py", line 759, in get_install_actions
    raise DependencyNeedsBuildingError(exc, subdir=subdir)
conda_build.exceptions.DependencyNeedsBuildingError: Unsatisfiable dependencies for platform linux-64: {'python=3.8[build=*_cpython]', 'pcl', 'pivy', 'doxygen', 'coin3d', 'smesh', 'pyside2', 'hdf5=1.10.6', 'boost-cpp=1.72', 'occt=7.4', 'libspnav', 'qt=5.12', 'xorg-libxt', 'netgen'}
```

Please let me know how to fix this problem. 

Thanks

looooo, 2020-05-08:
most likely you need to add the `conda-forge` channel:
`conda config --add channels conda-forge`

amrit3701, 2020-05-08:
@looooo 

After adding `conda-forge` channel, dependency error is fixed. But now error in compiling stage fails.

```
-- Build files have been written to: $SRC_DIR/build
FREECAD_USE_3DCONNEXION=
ninja: error: '/usr/lib64/libXext.so', needed by 'Mod/Fem/Fem.so', missing and no known rule to make it
```

#### Logs: 

```bash
ambu@ambu-pc:~/Desktop/freecad_build/freecad-feedstock$ conda build . -m .ci_support/linux_python3.8.____cpython.yaml 
No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
WARNING:conda_build.metadata:No numpy version specified in conda_build_config.yaml.  Falling back to default numpy value of 1.11
Adding in variants from internal_defaults
INFO:conda_build.variants:Adding in variants from internal_defaults
Adding in variants from /home/ambu/Desktop/freecad_build/freecad-feedstock/recipe/conda_build_config.yaml
INFO:conda_build.variants:Adding in variants from /home/ambu/Desktop/freecad_build/freecad-feedstock/recipe/conda_build_config.yaml
Adding in variants from .ci_support/linux_python3.8.____cpython.yaml
INFO:conda_build.variants:Adding in variants from .ci_support/linux_python3.8.____cpython.yaml
Attempting to finalize metadata for freecad
INFO:conda_build.metadata:Attempting to finalize metadata for freecad
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done
BUILD START: ['freecad-0.19.pre-py38hdf1be2e_117.tar.bz2']
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done

## Package Plan ##

  environment location: /home/ambu/miniconda3/conda-bld/freecad_1588956647943/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_plac

The following NEW packages will be INSTALLED:

    _libgcc_mutex:    0.1-conda_forge            conda-forge
    _openmp_mutex:    4.5-0_gnu                  conda-forge
    boost-cpp:        1.72.0-h8e57a91_0          conda-forge
    bzip2:            1.0.8-h516909a_2           conda-forge
    ca-certificates:  2020.4.5.1-hecc5488_0      conda-forge
    certifi:          2020.4.5.1-py38h32f6830_0  conda-forge
    coin3d:           4.0.0-hd7203a3_1009        conda-forge
    curl:             7.69.1-h33f0ec9_0          conda-forge                                                                                                                                 
    dbus:             1.13.6-he372182_0          conda-forge                                                                                                                                 
    doxygen:          1.8.18-hd1b7508_0          conda-forge                                                                                                                                 
    eigen:            3.3.7-hc9558a2_1001        conda-forge                                                                                                                                 
    expat:            2.2.9-he1b5a44_2           conda-forge                                                                                                                                 
    flann:            1.9.1-h44f99b7_1008        conda-forge                                                                                                                                 
    fontconfig:       2.13.1-h86ecdb6_1001       conda-forge                                                                                                                                 
    freeimage:        3.18.0-he1b5a44_0          conda-forge                                                                                                                                 
    freetype:         2.9.1-he983fc9_1006        conda-forge
    future:           0.18.2-py38h32f6830_1      conda-forge
    gettext:          0.19.8.1-hc5be6a0_1002     conda-forge
    glib:             2.64.2-h6f030ca_0          conda-forge
    gst-plugins-base: 1.14.5-h0935bb2_2          conda-forge
    gstreamer:        1.14.5-h36ae1b5_2          conda-forge
    hdf4:             4.2.13-hf30be14_1003       conda-forge
    hdf5:             1.10.6-nompi_h3c11f04_100  conda-forge
    icu:              64.2-he1b5a44_1            conda-forge
    jpeg:             9c-h14c3975_1001           conda-forge
    jsoncpp:          1.8.4-hc9558a2_1002        conda-forge
    krb5:             1.17.1-h2fd8d38_0          conda-forge
    ld_impl_linux-64: 2.34-h53a641e_0            conda-forge
[... 740 lines cut ...]
-- Generating done
-- Build files have been written to: $SRC_DIR/build
FREECAD_USE_3DCONNEXION=
ninja: error: '/usr/lib64/libXext.so', needed by 'Mod/Fem/Fem.so', missing and no known rule to make it
Traceback (most recent call last):
  File "/home/ambu/miniconda3/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/cli/main_build.py", line 469, in main
    execute(sys.argv[1:])
[... cut ...]
    return _func_defaulting_env_to_os_environ('call', *popenargs, **kwargs)
  File "/home/ambu/miniconda3/lib/python3.7/site-packages/conda_build/utils.py", line 378, in _func_defaulting_env_to_os_environ
    raise subprocess.CalledProcessError(proc.returncode, _args)
subprocess.CalledProcessError: Command '['/bin/bash', '-o', 'errexit', '/home/ambu/miniconda3/conda-bld/freecad_1588956647943/work/conda_build.sh']' returned non-zero exit status 1.
ambu@ambu-pc:~/Desktop/freecad_build/freecad-feedstock$ 

```

amrit3701, 2020-05-08:
@looooo Do I have to install some additional dependencies for `Fem.so`?

looooo, 2020-05-08:
looks good. Normally conda-packages are build on a centos6 docker. In ubuntu you can link `/usr/lib/x86_64-linux-gnu/` to `/usr/lib64`. `ln -ls /usr/lib/x86_64-linux-gnu /usr/lib64`
if you want to build with docker, look here for instruction:
https://github.com/FreeCAD/FreeCAD_Conda/blob/master/build.md#linux

amrit3701, 2020-05-08:
Very much thanks @looooo 

I tried to build FreeCAD in docker as you said and successfully able to create AppImage. 

The `linux_dev.sh` fails on bundling AppImage with error `dlopen(): error loading libfuse.so.2`
Then on searching, I found that I have to add `--appimage-extract-and-run` flag to command. (https://github.com/AppImage/AppImageKit/wiki/FUSE#docker)

```bash
ARCH=x86_64 ../../appimagetool-x86_64.AppImage --appimage-extract-and-run \
  -u "gh-releases-zsync|FreeCAD|FreeCAD|$DEPLOY_RELEASE|FreeCAD*glibc2.12-x86_64.AppImage.zsync" \
  AppDir  ${version_name}.AppImage
```

For building FreeCAD on Ubuntu, I also create logical symbolic link (`ln -ls /usr/lib/x86_64-linux-gnu /usr/lib64`) and still getting same error (https://github.com/FreeCAD/FreeCAD_Conda/issues/38#issuecomment-625918819). I would like to know the exact reason behind this issue and how to fix.

Anyways, can close this ticket.

Thanks again.

looooo, 2020-05-09:
> For building FreeCAD on Ubuntu, I also create logical symbolic link (ln -ls /usr/lib/x86_64-linux-gnu /usr/lib64) and still getting same error (#38 (comment)). I would like to know the exact reason behind this issue and how to fix.

Normally conda should not link against libraries in /usr/lib... . There is the solution with cdt's:
https://docs.conda.io/projects/conda-build/en/latest/resources/define-metadata.html
>On Linux, using the compiler packages provided by Anaconda Inc. in the defaults meta-channel can prevent your build system leaking into the built software by using our CDT (Core Dependency Tree) packages for any "system" dependencies. These packages are repackaged libraries and headers from CentOS6 and are unpacked into the sysroot of our pseudo-cross compilers and are found by them automatically.

But this does not work for all recipes. Not sure which package needs the `/usr/lib64/libxext.so`. To improve this situation it might be interesting to find out which package contains this path.

As a workaround you need to provide this file somehow. Look for `libxext.so.` in `/usr/lib/x86_64-linux-gnu/`. If it is available creating a soft link from `/usr/lib/x86_64-linux-gnu/` to `/usr/lib64` should be sufficient. Btw.: I made an error in my suggested command. It should be `ln -s /usr/lib/x86_64-linux-gnu /usr/lib64`

amrit3701, 2020-05-13:
Hi @looooo 

Now, I am compiling FreeCAD on MacOS to make *.dmg file. 

#### Steps I followed:
1. Install Minicoda (https://docs.conda.io/en/latest/miniconda.html) Python3.7
1. Followed three steps mentioned https://github.com/FreeCAD/FreeCAD_Conda/blob/master/build.md#macosx
    * Download Download the MacOSX10.12.sdk
    * Extract files to `/opt/MacOSX10.12.sdk`
    * Add a file in the user home directory AKA ~/conda_build_config.yaml containing these lines:
    ```
    CONDA_BUILD_SYSROOT:  
        - /opt/MacOSX10.12.sdk        # [osx]
    ```
1. Cloned `freecad-feedstock`
1. Run `sudo conda build . -m .ci_support/osx_python3.8.____cpython.yaml`

Getting error:

```bash
adding hacks for osx
ln: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk: No such file or directory
Traceback (most recent call last):
  File "/Users/ambu/miniconda3/bin/conda-build", line 11, in <module>
    sys.exit(main())
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/cli/main_build.py", line 474, in main
    execute(sys.argv[1:])
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/cli/main_build.py", line 465, in execute
    verify=args.verify, variants=args.variants)
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/api.py", line 209, in build
    notest=notest, need_source_download=need_source_download, variants=variants)
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/build.py", line 2864, in build_tree
    notest=notest,
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/build.py", line 1995, in build
    cwd=src_dir, stats=build_stats)
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/utils.py", line 404, in check_call_env
    return _func_defaulting_env_to_os_environ('call', *popenargs, **kwargs)
  File "/Users/ambu/miniconda3/lib/python3.7/site-packages/conda_build/utils.py", line 384, in _func_defaulting_env_to_os_environ
    raise subprocess.CalledProcessError(proc.returncode, _args)
subprocess.CalledProcessError: Command '['/bin/bash', '-o', 'errexit', '/Users/ambu/miniconda3/conda-bld/freecad_1589374927662/work/conda_build.sh']' returned non-zero exit status 1.
```
On debugging, I found the script fails `conda_build.sh` (https://paste.ubuntu.com/p/Dz3p52nYMv/) at line no 29 (create symbolic link).

As I don't have XCode then I installed XCode from AppStore, 
```
λ /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/ ls
DriverKit19.0.sdk MacOSX.sdk        MacOSX10.15.sdk
```

In the symbolic link you mentioned to `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk` file points to `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk` and in the XCode I `MacOSX10.12.sdk` is not present. There is `MacOSX10.15.sdk`.

Do, I need the change the `ln` command?

[build_env_setup.sh](https://paste.ubuntu.com/p/mx34Q9hsn7/)

Here are the full logs of `sudo conda build . -m .ci_support/osx_python3.8.____cpython.yaml` commands:
https://paste.ubuntu.com/p/4w9b7trh6g/

looooo, 2020-05-13:
I guess you need macosx10.12.sdk. you can try with 10.15 but 1. we never tested with 10.15 and 2. all conda-forge packages are build with 10.9 , 10.12.

amrit3701, 2020-05-13:
> I guess you need macosx10.12.sdk. you can try with 10.15 but 1. we never tested with 10.15 and 2. all conda-forge packages are build with 10.9 , 10.12.

Yes, but I already put macos10.12.sdk in `/opt/` directory. Also in [build_env_setup.sh](https://paste.ubuntu.com/p/mx34Q9hsn7/) file we have 
`export CONDA_BUILD_SYSROOT="/opt/MacOSX10.12.sdk"` at line no. 56

Is condo took SDK's from Xcode.app? Or should I copy `MacOSX10.12.sdk` from `/opt` to `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/`?

looooo, 2020-05-13:
Maybe this helps:
https://github.com/FreeCAD/FreeCAD_Conda/blob/master/build.md#macosx

amrit3701, 2020-05-15:
> Maybe this helps:
> https://github.com/FreeCAD/FreeCAD_Conda/blob/master/build.md#macosx

@looooo Yes, it works. 👍 No need to install XCode as condo is using sdk from `CONDA_BUILD_SYSROOT`.

Thanks a lot!

closed 2020-05-15 by amrit3701