FreeCAD/FreeCAD_Conda #40: Failed: conda build . -m .ci_support/linux_python3.8.____cpython.yaml

amrit3701, 2020-07-16:
Hi @looooo 

I am creating an AppImage for Linux using docker and getting below errors:

Logs of `conda build . -m .ci_support/linux_python3.8.____cpython.yaml`: https://paste.ubuntu.com/p/sk2DK3BKv4/

CMakeError.log: https://paste.ubuntu.com/p/rChRWNYBGJ/

From errors, it looks like CMake unable to found lpthreads. Do, I have to create any symbolic link (`ln -s`) inside a docker container.

### Steps I followed: 

1. Clone https://github.com/looooo/freecad-feedstock.
1. Run `sudo docker run -i -t -v ~/projects/:/home/conda/projects  --name conda-forge condaforge/linux-anvil-comp7` command. Here in `~/projects/` folder I have freecad-feedstock.
1. After entering into docker shell, I run `/usr/bin/sudo yum install -y mesa-libGL-devel` command.
1. Also, run `conda update --all` command.
1. Go the freecad-feedstock directory (`cd ~/projects/freecad-feedstock`).
1. Run `conda build . -m .ci_support/linux_python3.8.____cpython.yaml` command. 

Source: https://github.com/FreeCAD/FreeCAD_Conda/blob/master/build.md#linux

On debugging, I found that process failed at `cmake` command. (https://github.com/looooo/freecad-feedstock/blob/master/recipe/build.sh#L38)

Before three months, I was successfully able to create an AppImage (https://github.com/FreeCAD/FreeCAD_Conda/issues/38) by following same above procedure.

Thanks in advance!

looooo, 2020-07-17:
until now we didn't run into this issue. Let's see if we can reproduce on the next full build.

looooo, 2020-07-20:
https://github.com/looooo/freecad-feedstock/commits?author=sgrogan
doesn't seem like we run into this with the ci-services.

amrit3701, 2020-07-20:
@looooo okay, I will re-try it again and let you know. thanks.