FreeCAD/FreeCAD_Conda #41: bump occt to 7.5.1

luzpaz, 2021-02-13:
https://anaconda.org/conda-forge/occt
7.5.1 adds error checking that 7.4.0 is missing and makes FC hang
Thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=55523&p=477790#p477783

luzpaz, 2021-02-14:
ping @looooo @sgrogan

looooo, 2021-02-15:
I still would like to go to vtk9 first. And do the occt 7.5 migration later.

gbroques, 2021-03-23:
This is also important and desirable for `.gltf` export:
https://forum.freecadweb.org/viewtopic.php?f=8&t=31680&p=450917&#p450658

See Also:
https://www.khronos.org/gltf/

looooo, 2021-03-23:
@luzpaz @gbroques we are only waiting for gmsh to be updated to occt7.5:
https://github.com/conda-forge/gmsh-feedstock/pull/39
https://forum.freecadweb.org/viewtopic.php?f=42&t=56963&p=490073#p490073

luzpaz, 2023-02-01:
[image]

This ticket can be closed.

closed 2023-02-01 by luzpaz