FreeCAD/FreeCAD_Conda #42: install FreeCAD_Conda inside an existing environment

hex41434, 2021-03-05:
Hi, 

thanks for your previous help, I could successfully create a new <freecad_fem_env> environment and execute my codes on Windows, Mac and Linux. 

now my question: I do not know how can I add this package to an existing environment (for instance : running freecad scripts (including fem) inside my existing environment along with Pytorch )

(as a second solution, I think I can create a freecad env and then add Pytorch to it - but I still prefer to use my existing env because I have a lot of dependencies there)

Thanks In advance

looooo, 2021-03-05:
You can simple use `conda install freecad` within the specific environment. Otherwise you can create a new environment with multiple does like this:
`conda create -n my_env freecad pytorch ...`

hex41434, 2021-03-05:
I actually tried this:
conda install freecad qt=5.9.* -c freecad/label/dev -c conda-forge 

and unfortunately, it has conflict with my Linux: (I have cuda 10.1)

[image]

could you recommend me an specific version of freecad that FEM toolbox works well and has no conflict with my environment?

looooo, 2021-03-05:
Why at.5.9 ? Recent freecad uses qt5.12.

hex41434, 2021-03-09:
I have a Linux OS and cuda: version 10.1
I tried to install a clean conda environment for pytorch using [this link](https://pytorch.org/get-started/previous-versions/): 
**conda install pytorch==1.7.1 torchvision==0.8.2 torchaudio==0.7.2 cudatoolkit=10.1 -c pytorch**
and then 
**conda install -c conda-forge freecad**
and I got this: 

[image]

I would be happy if you have any suggestion ( I need FEM package besides Pytorch)